weigao95/surfelwarp #3: The pre-allocated buffer is not enough.

NuageLee, 2019-09-10:
Hi,
I run it on Ubuntu16.04 with GTX 980ti, it says "The pre-allocated buffer is not enough" and give me a runtime_error.
buffer size = 50000 and size = 72748. 

Any help will be much appreciated.

weigao95, 2019-09-10:
You might modify the constants in ${project_root}/common/Constants.cpp

50000 seems to be "Constants::kMaxNumSurfelCandidates", you can change it to a larger number like 100000.

closed 2019-09-23 by weigao95

ZYJxxx, 2021-03-29:
> Hi,
> I run it on Ubuntu16.04 with GTX 980ti, it says "The pre-allocated buffer is not enough" and give me a runtime_error.
> buffer size = 50000 and size = 72748.
> 
> Any help will be much appreciated.

hi,
have you solved this problem, and how did you do that? hope for your reply.