weigao95/surfelwarp #4: Get rid of PCL dependency

Algomorph, 2019-09-10:
This is, in a sense, a feature request, or, to be more exact, an enhancement request.

PCL's versions are not very backward-compatible, and building just the right PCL with just the right feature support means messing with the dependency hell so very inherent to PCL.

To give some idea, just to deal with the old PCL version + support for only the older version of CUDA already took me 6 over hours to resolve, and I'm not done yet.

My recommendation is to keep this issue open even if the maintainers decide they don't want to address this right now. Maybe, someone will come up with a PR, IDK.

weigao95, 2019-09-10:
Yeah, the code is developed when CUDA 9 is the latest. PCL is used mainly for debug visualization and it might worth to keep it only in `${project_root}/visualization` and mark as optional.

I will take a look after the ICRA deadline and you are very welcomed to come up with a PR if you want.

weigao95, 2019-09-22:
Thanks for your work, that's great! I will test it on my machine.

PR #5 (2019-09-23): Add support for cilantro to optionally replace PCL requirement
- Automatically extends support to CUDA 10.X, VisualStudio 2017, and Ubuntu 18.04
- Not tested in VisualStudion 2015 or on Ubuntu 16.04
- Not fully tested with PCL build (due to compatibility issue nightmare on my available platforms), but should still work in theory
- Tested with CUDA 9.0
- Fixed Boost linkage on Windows in corner cases
- FindGLFW.cmake script was dropped -- the glfw3 package provides its own cmake module, i.e. glfw3Config.cmake, which should be used on both platforms instead of the non-standard module.
- Added CM [... cut]
closed 2019-09-23 by weigao95