weigao95/surfelwarp #6: Add AppVeyor CI

Algomorph, 2019-09-19:
This is another enhancement request.
It naturally flows out of #5 if the PR is accepted.

In order to properly maintain this package and preempt any PR from breaking core functionality on any platform, I propose to add AppVeyor testing to this repository.

For the initial build tests, I propose the following build configuration matrix:

- [ ] Microsoft Visual Studio 2015, PCL 1.8, CUDA 9.2
- [ ] Microsoft Visual Studio 2017, cilantro, CUDA 9.2 
- [ ] Microsoft Visual Studio 2017, cilantro, CUDA 10.1
- [ ] Ubuntu 16.04 / gcc 5.3, PCL 1.8, CUDA 9.2
- [ ] Ubuntu 18.04 / gcc 7.4, cilantro, CUDA 10.1

weigao95, 2019-09-23:
I've tested Visual Studio 2016 and ubuntu with CUDA 9.2 and PCL. After some minor fix, it works well.

Adding CI would be great, but to be honest I haven't work with it before... Our research group uses CI for several projects and I may need to learn from experienced colleagues. 

BTW, I'm sorry that I accidentally push to your master when fixing build with PCL. I didn't realize that I got write access to the fork automatically.

Algomorph, 2019-09-23:
@weigao95 , 
Firstly, your fixes to the master branch of my fork are more than welcome, and thanks for doing that.

As to CI, I'll need a few more days to determine whether I should continue working on your codebase or not in the short term. If I come to the decision to proceed, I will tackle the basic set up with AppVeyor and the cilantro-based tests. From that, you would have enough material to easily add the two tests with PCL into the matrix. We'll see how this goes.

Algomorph, 2019-10-24:
Just a brief update on this. I have pretty much figured out how to test CUDA-based C++ project in appveyor working on my own project. However, before I can finally test completely, I need to get rid of one dependency that's slowing down my repo's build too much. This task is in my queue, as soon as I finish and test, I can take a look at some basic appveyor support here.

weigao95, 2019-10-25:
That would be great! Thanks for your work!