weigao95/surfelwarp #7: Simplify the dependency building using CMake's ExternalProject_Add

Algomorph, 2019-09-19:
Using CMake's [ExternalProject_Add](https://cmake.org/cmake/help/latest/module/ExternalProject.html) to check-out and automatically build dependencies can probably dramatically simplify the build process, esp. on the Windows platform, where more things have to be built from scratch. This would thereby aid in the adoption of SurfelWarp by the community.

Hence I propose to have corresponding CMake options, i.e.
BUILD_EXTERNAL_GLFW, which would trigger the download and build using CMake subroutines (which use git). Otherwise, the system version of the package would be used.

Refer to [Pangolin](https://github.com/stevenlovegrove/Pangolin) for reference.

Again, if #5 is accepted, these can be the target dependencies:
- [ ] Eigen
- [ ] GLFW
- [ ] Pangolin
- [ ] cilantro

I wouldn't include OpenCV or PCL here because those are massive and require eons to build, esp. with common dependencies. Moreover, OpenCV is pre-packaged and available for both Windows and Ubuntu.

This addition would also allow not to include Eigen with the code but get it from a specific commit, thereby reducing the size of the repository and amount of clutter.

wangmiaowei, 2020-12-07:
@weigao95 @Algomorph When I run ./surfelwarp_app /test_data/boxing_config.json I get an error Segmentation fault (core dumped)
I do not know the reason.

Algomorph, 2020-12-08:
@weigao95 I'm not using this specific codebase right now. It may have bugs stemming from dependency upgrades or some specific setup on the system. It took me two weeks of coding just to get it working, but hopefully it will be much easier for you with my cilantro integration. 

If you have very specific issues, e.g. determine which line of the code segfaults, try to figure out what part of the code or third-party call is failing, maybe I could help you.

Algomorph, 2020-12-08:
On the bright side, I've figured out how to do integration testing with CUDA, but for now I'm still using BOYC for Appveyor to do this for the codebase I'm working on right now (because the build & testing are so slow and it times out).

If you really want to ensure your code keeps working for other people, I recommend investing some time in exploring continuous integration. That's the only real way to ensure things continue working while third-party libraries are upgraded. Bear in mind that the whole reason for my cilantro integration was that it was virtually impossible to get the right combination of severely outdated PCL + other libraries back when I was trying your code out.

Algomorph, 2020-12-08:
Also, @wangmiaowei , you should open a separate issue if you're experiencing an unrelated build problem.