weigao95/surfelwarp #16: The pre-allocated buffer is not enough

AzureSilent, 2019-11-08:
Hello!
When I run the code with the data-set **boxing** , it failed with:
`surfelwarp/common/DeviceBufferArray.h:90: The pre-allocated buffer is not enough`
(m_surfel_candidate_pixel size = 72748  >  m_surfel_vertex_confid buffer = 50000 )

this happened after the 
surfelwarp_app/main.cpp:40: The 0th Frame
then : fusion.ProcessNextFrameWithReinit(offline_rendering)
then: m_live_geometry_updater->ProcessFusionStreamed(num_remaining_surfel, num_appended_surfel);

and then, I found that in 
surfelwarp::SurfelFusionHandler::processFusionAppendAtomic()
it calculates the wrong result： 
the m_atomic_appended_pixel_index was set to 72748

could you or any one please help me find out where is wrong?

weigao95, 2019-11-12:
Sorry for my delayed reply as the notification email from github were filtered.

I've re-run the boxing sequence on my machine with the default config, and it works well. Did you modify the configuration file? If so, please send a copy to my email and I will have a try.

If you are using the default config, could you make sure you have compiled the project with the correct cuda arch flag? What's GPU are you using?

AzureSilent, 2019-11-13:
Thanks for your reply.

I didn't change anything. I'm testing on GTX 670(2G) with arch set to 30 which is according to the official website。
I think this is caused by the memory allocation.  
I have allocated the error:
when the program processes at function：
fusion.ProcessNextFrameWithReinit(offline_rendering);  （the 0 frame）
the following step in this function goes wrong：
 m_image_processor->ProcessFrameStreamed(observation, m_frame_idx); (/surfelwarp/core/SurfelWarpSerial.cpp )
in the right circumstance, it will set the value of  m_warp_field->SolverAccess() ->node_se3 to something like [[0,0,0,1]...] 
but if it goes wrong , the value  will be random value.
_____________________________
1.if the number of downsampled valid surfels from the 0th img is bigger than 65000, this error usually happens, but 2G ram is enough, because when it goes well, the ram consumption is around 1.5G.
further, as the program goes on in the right scenario, the surfels can up to 150000,  and it still goes well.

weigao95, 2019-11-13:
I'm little bit confused: Is that the problem happens when the number of downsampled valid surfels from the 0th img is bigger than 65000?

If the problem is related to the initialization of warp field, can you check the output of [this line](https://github.com/weigao95/surfelwarp/blob/ea136390befb6d87c5e083c969c9c7cb453f893d/core/SurfelWarpSerial.cpp#L83) to see whether the `m_warp_field` is set to some random value.

Sorry but I don't have a GPU similar to yours and cannot reproduce the problem by myself.

AzureSilent, 2019-11-15:
No, If this happened, the code just stoped.
But if I some how change the input images to just let the valid surfels from the 0th img smaller than 65000, your code gose well.
I have debug the code, it just happened at [here](https://github.com/weigao95/surfelwarp/blob/ea136390befb6d87c5e083c969c9c7cb453f893d/core/SurfelWarpSerial.cpp#L127). It seems all things goes well before this funcion. but this function set the value of m_warp_field->SolverAccess() ->node_se3 form  [[0,0,0,1]...] to random values.

————————————————
by the way, I have another question.
I found this code didn't display anything, could I ger ride of the opengl？ so I can test the code on my server which has much better GPU card.
my server doesn't have any display device. So I can only use softwares like X server to test.  I have tried many many methods, but still get the GL version limited to 3.0.

Why your code needs at least GL version 4.5, could I avoid this？

weigao95, 2019-11-15:
It is strange because the code you mentioned doesn’t do anything to warp field. If the problem is with warp field, the bug should be located in `ProcessFirstFrame` instead of `ProcessNextFrameWithInit`.

OpenGL is used to render the index map and solver/fusion maps, which are quite essential to the pipeline. Thus, I don’t think it is easy to get rid of OpenGL.

There is a big change in OpenGL 3.3 (if I remember it correctly) where the programmer can access lower-level APIs. OpenGL 4.5 is not strict, but I think the GL version should be above 3.3

weigao95, 2019-11-18:
To be honest I'm not a expert on GPU server and not able to help. I'm afraid the setup effort is not trivial.

I took a look at EGL. That seems to be a completely different setup of graphic API, which requires at least re-write the core/render module, which is a huge amount of work (and I don't know would it work with CUDA).

AzureSilent, 2019-11-19:
It's a bad news...  T_T
I'll try other ways...
Anyway, thank you.

closed 2019-11-20 by weigao95

ZYJxxx, 2021-03-12:
> No, If this happened, the code just stoped.
> But if I some how change the input images to just let the valid surfels from the 0th img smaller than 65000, your code gose well.
> I have debug the code, [...]

I had the same problem, "The pre-allocated buffer is not enough", Have you solved it yet? How did you solve it?

prl666, 2021-04-02:
@weigao95 @AzureSilent I had the same problem, when I set the kMaxNumSurfelCandidates = 100000 ， it fails with "Nan in node candidate", Have you solved it yet? How did you solve it? My system if ubuntu 18.04 cuda10.0 GTX970.
plsease help me ,thanks！