weigao95/surfelwarp #17: make error: "Matrix3f" is undefined

plutoyuxie, 2019-11-12:
The error is related to Eigen I think, and I change the common_types.h by
#include <external/eigen3.4/Eigen/Eigen>
but it still doesn't work. Hope for some help, thanks.  

Logs:

-- Build files have been written to: /home/surfelwarp/build
[  1%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_common_utils.cu.o
/home/surfelwarp/common/common_types.h(25): error: identifier "Matrix3f" is undefined

/home/surfelwarp/common/common_types.h(25): error: expected a ";"

/home/surfelwarp/common/common_types.h(26): error: identifier "Vector3f" is undefined

/home/surfelwarp/common/common_types.h(26): error: expected a ";"

/home/surfelwarp/common/common_types.h(27): error: identifier "Matrix4f" is undefined

/home/surfelwarp/common/common_types.h(27): error: expected a ";"

/home/surfelwarp/common/common_types.h(28): error: identifier "Vector4f" is undefined

/home/surfelwarp/common/common_types.h(28): error: expected a ";"

/home/surfelwarp/common/common_types.h(29): error: identifier "Matrix6f" is undefined

/home/surfelwarp/common/common_types.h(29): error: expected a ";"

/home/surfelwarp/common/common_types.h(30): error: identifier "Vector6f" is undefined

/home/surfelwarp/common/common_types.h(30): error: expected a ";"

/home/surfelwarp/common/common_types.h(31): error: identifier "MatrixXf" is undefined

/home/surfelwarp/common/common_types.h(31): error: expected a ";"

/home/surfelwarp/common/common_types.h(32): error: identifier "VectorXf" is undefined

/home/surfelwarp/common/common_types.h(32): error: expected a ";"

/home/surfelwarp/common/common_types.h(33): error: identifier "Isometry3f" is undefined
......

JenningsL, 2019-11-12:
Switching to gcc-7 solved this problem for me.

plutoyuxie, 2019-11-13:
> Switching to gcc-7 solved this problem for me.

Thanks a lot! @JenningsL This issue is solved and gcc-6 is suitable for me.

make is done but new error occurs when I run ./surfelwarp_app,  logs as below:

/3d/code/surfelwarp/core/render/Renderer.cpp:17: The graphic pipeline is not correctly initialized

It is due to glfwInit() failure. I have no idea what happen. Hope for helps again.

plutoyuxie, 2019-11-13:
> Switching to gcc-7 solved this problem for me.

Thanks a lot! @JenningsL This issue is **solved** and gcc-6 is suitable for me.

make is done but new error occurs when I run ./surfelwarp_app, logs as below:
/3d/code/surfelwarp/core/render/Renderer.cpp:17: The graphic pipeline is not correctly initialized
It is due to **glfwInit() failure**. I have no idea what happens. Hope for help again.

weigao95, 2019-11-13:
You can try a simpler GLFW applications, for instance [this one](https://www.glfw.org/docs/latest/quick.html) on its website to see if you can call glfwInit correctly and create a window.

BTW, what's your GPU and driver version?

plutoyuxie, 2019-11-14:
> You can try a simpler GLFW applications, for instance this one on its website to see if you can call glfwInit correctly and create a window.
> BTW, what's your GPU and driver version?

Hi, thank you for the reply. I use 2080ti and cuda 10.

closed 2019-11-14 by plutoyuxie

Cartucho, 2020-05-04:
I tried to compile this repo on two different computers and I get the same error.

Computer 1: NVIDIA 20800 ti, CUDA 10.2, Ubuntu 16.04
Computer 2: NVIDIA GEFORCE GTX 950M, Cuda 9.0, Ubuntu 16.04

Cartucho, 2020-05-04:
If you are using CUDA 10, switching to gcc-7 like @JenningsL suggested indeed works.

Cartucho, 2021-05-09:
As explained [here](https://github.com/mp3guy/ICPCUDA/issues/30), this will happen in Ubuntu due to the low version of c++ during compiling. You can solve it by flagging the compiler to C++11 (`-std=c++11`):  https://github.com/weigao95/surfelwarp/blob/5e30876c97840f24ab736ea6e435619f79cc907d/CMakeLists.txt#L70