weigao95/surfelwarp #18: token __CUDACC_VER__ is no longer supported

JenningsL, 2019-11-12:
Hi, thanks for your code. I install the eigen in /external folder, and install cuda9.0. I tried  to compile the code following the readme. However, I got the following error: 

`[  1%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o
In file included from /usr/local/cuda/include/common_functions.h:50:0,
                 from /usr/local/cuda/include/cuda_runtime.h:115,
                 from <command-line>:0:
/usr/local/cuda/include/crt/common_functions.h:64:24: error: token ""__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."" is not valid in preprocessor expressions
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^
/usr/include/eigen3/Eigen/src/Core/util/Macros.h:402:85: note: in expansion of macro ‘__CUDACC_VER__’
   && ( !defined(__NVCC__) || !EIGEN_ARCH_ARM_OR_ARM64 || (defined __CUDACC_VER__ && __CUDACC_VER__ >= 80000) )
`

It seems to be the incompatity between Eigen and CUDA, however, I don't know exactly how to fix it. A little help would be appreciated. Thanks.

JenningsL, 2019-11-12:
Closed. It turn out that I am still using the system installed eigen.

closed 2019-11-12 by JenningsL

xiaotaw, 2020-02-17:
Hi, @JenningsL. I come across the same problem, could you share the Eigen version which is compatible with CUDA=9.0?

My system installed eigen = 3.3.4

Cartucho, 2021-05-09:
@xiaotaw you should set the Eigen path to the one that is provided in this GitHub repo, you can find it in `surfelwarp/external/
eigen3.4`

xiaotaw, 2021-05-11:
> @xiaotaw you should set the Eigen path to the one that is provided in this GitHub repo, you can find it in `surfelwarp/external/ eigen3.4`

Yeah, thank you. I have solved this problem by the way you said.