weigao95/surfelwarp #22: CUDA_cublas_device_LIBRARY-NOTFOUND

OptSolution, 2019-11-22:
I use cuda 10.1 on ubuntu 18.04, when I cmake

I get the error 

`CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_cublas_device_LIBRARY (ADVANCED)`

```
$ nvcc -V                         
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2019 NVIDIA Corporation
Built on Wed_Apr_24_19:10:27_PDT_2019
Cuda compilation tools, release 10.1, V10.1.168
```

weigao95, 2019-11-22:
What's the version of cmake you are using? Can you try a higher version?

OptSolution, 2019-11-25:
The version in the cmake-gui is 3.10.2, I update cmake to 3.15.5 it works. Thanks

closed 2019-11-25 by OptSolution