weigao95/surfelwarp #23: a error when running the demo

Cryst4L9527, 2019-12-22:
when I‘m running the demo under cuda9.1 and cilentro on 2080Ti，there is a error at line534 of SurfelFusionHandler.cu，It seems that there is something wrong about the cudaMemcpyAsync.Does anyone know why？thx！

Cryst4L9527, 2019-12-22:
I've printed the error message:
an illegal memory access was encountered.
It's just the original code I'm running.

Cryst4L9527, 2019-12-22:
and I've try a simple sample in my environment,the function cudaMemcpy and cuda Malloc are running in a right way.

weigao95, 2019-12-23:
I am not familiar with the cilentro environment (I use pcl instead), but I remembered this problem happened when the cuda arch flag is not correct. Can you check that?

Cryst4L9527, 2019-12-24:
I run it on a RTX 2080Super and set the cuda arch flag as 75,and run under cuda 10.0.Is there anything wrong?
I even change the CMakeList.txt as following:

# Cuda
find_package(CUDA 10 REQUIRED)
include_directories(${CUDA_INCLUDE_DIRS})
set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -arch=sm_75")
set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -gencode=arch=compute_75,code=sm_75")
set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -gencode=arch=compute_75,code=compute_75")

but it still doesn't work.

weigao95, 2019-12-26:
Sorry but I don't have cuda10 environment and cannot reproduce the error. Thus, I can only provide you some advice.

Can you set the `CUDA_DEBUG_SYNC_CHECK` flag to 1 in `common/global_configs.h` and run the code again?

Due to the sync mechanism of cuda kernel, the error might not be caused by the code that throws it (line534 in your case). The most common case is the kernel is launched with incorrect parameters (for instance, zero in block size), but the exception will be throw only when `cudaSafeCall` is invoked.

closed 2020-01-02 by weigao95

Cryst4L9527, 2020-01-05:
I set the flag and the error changed:
/home/cscg/surfelwarp/core/geometry/SurfelNodeDeformer.cu:147

#if defined(CUDA_DEBUG_SYNC_CHECK)
	cudaSafeCall(cudaStreamSynchronize(stream));//here
	cudaSafeCall(cudaGetLastError());
#endif

Is this error familiar?^-^

weigao95, 2020-01-06:
Can you check device::forwardWarpVertexAndNodeKernel ([this line](https://github.com/weigao95/surfelwarp/blob/82f1d25e10490e97460d3a21c853eb98c234d92f/core/geometry/SurfelNodeDeformer.cu#L130)) to make sure its input parameter is reasonable (the size is correct and contains non NaN)?