weigao95/surfelwarp #25: How to compile `pangolin` with `Eigen`?

Wjiajie, 2020-01-12:
I try to compile `surfelwarp` on win10 ,with vs2017 ,cuda 10.1 and `cilantro`,when i compile `cilantro`,i got an error:`Namespace "pangolin" has no member "ToEigen"`。But i fail to compile `pangolin with eigen3`,I set `BUILD_PANGOLIN_EIGEN` to be`true`,and also set `EIGEN_INCLUDE_DIR` to my Eigen's include folder,but i didnt work for me . Cmake Logs of `pangolin`,seems not mention `eigen` at all:
```
Selecting Windows SDK version 10.0.17763.0 to target Windows 10.0.18362.
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
Could NOT find PkgConfig (missing: PKG_CONFIG_EXECUTABLE) 
OpenNI2 Found and Enabled
MediaFoundation Found and Enabled
libpng Found and Enabled
libjpeg Found and Enabled
Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
Configuring done
```

I use latest version of Cmake, Eigen version is `Eigen 3.3.9`  included in `surfelwarp/external/eigen3.4`.

Wjiajie, 2020-01-15:
Oh~ my fault ，i have solved the problem...

closed 2020-01-15 by Wjiajie

weigao95, 2020-01-15:
Sorry for my delayed replay and happy to see you resolve this problem. The notification email from github was classified as spam and I have not seen it. You can send me an email if I don't reply for a long time.

Wjiajie, 2020-01-16:
Thanks for reply,so kind of you~~