weigao95/surfelwarp #26: Error building on Ubuntu 19.04

bsespede, 2020-01-22:
During building procedure I get these errors when building `Building NVCC (Device) object hashing/CMakeFiles/hash_lib.dir/hash_lib_generated_CompactProbeTable.cu.o
`:

`usr/include/eigen3/Eigen/src/Core/arch/CUDA/PacketMathHalf.h(81): error: no instance of overloaded function "__ldg" matches the argument list
            argument types are: (const Eigen::half *)

/usr/include/eigen3/Eigen/src/Core/arch/CUDA/PacketMathHalf.h(81): error: no instance of overloaded function "__ldg" matches the argument list
            argument types are: (const Eigen::half *)

2 errors detected in the compilation of "/tmp/tmpxft_00001adb_00000000-8_sanity_check.compute_61.cpp1.ii".
CMake Error at common_lib_generated_sanity_check.cu.o.cmake:279 (message):
  Error generating file
  /home/bsespede/Repositories/surfelwarp/build/common/CMakeFiles/common_lib.dir//./common_lib_generated_sanity_check.cu.o

make[2]: *** [common/CMakeFiles/common_lib.dir/build.make:93: common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o] Error 1
2 errors detected in the compilation of "/tmp/tmpxft_00001ae8_00000000-8_algorithm_types.compute_61.cpp1.ii".
CMake Error at common_lib_generated_algorithm_types.cu.o.cmake:279 (message):
  Error generating file
  /home/bsespede/Repositories/surfelwarp/build/common/CMakeFiles/common_lib.dir//./common_lib_generated_algorithm_types.cu.o

make[2]: *** [common/CMakeFiles/common_lib.dir/build.make:65: common/CMakeFiles/common_lib.dir/common_lib_generated_algorithm_types.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:108: common/CMakeFiles/common_lib.dir/all] Error 2` 

Could this error be related to my CUDA version (its 10.1)? Im using the prebuilt eigen version (libeigen3-dev).

Thanks!

weigao95, 2020-01-24:
That seems to be the problem of using Eigen in CUDA. Can you try to write a simple .cu file that include Eigen and make sure it works?

Sorry but I don't have Ubuntu 19 environment available.

bsespede, 2020-01-29:
It was Indeed an incompatibility issue between CUDA 10.2 and Eigen 3.7. Had to downgrade to CUDA 9.2.

Now I'm having issues with during linking with cilantro, as apparently the file <cilantro/point_cloud.hpp> was moved in recent versions. May I ask you which commit you used for cilantro? Otherwise I'll give PCL a go.

Thanks for your time!

bsespede, 2020-01-30:
Hi Wei,

I've finally managed to find a working cilantro version and it builds perfectly fine. However, after I downloaded and tried to run the sample 'boxing' scene I get the following error during runtime:

`/home/bsespede/Repositories/surfelwarp/cmake-build-debug/apps/surfelwarp_app/surfelwarp_app ./data/boxing_config.json
Error: 	/home/bsespede/Repositories/surfelwarp/common/algorithm_types.h:85`

After checking that file, it seems to crash during the construction of `FlagSelection`, so it seems CUDA related? I'm using CUDA 9.2 for reference.

closed 2020-01-30 by bsespede