weigao95/surfelwarp #28: Some cilantro files have moved

timhutton, 2020-01-30:
e.g. the SurfelWarp code refers to `#include <cilantro/point_cloud.hpp>` which doesn't exist - is now `cilantro/utilities/point_cloud.hpp`

bsespede, 2020-01-31:
Hi Tim, I've used this commit from cilantro and it worked perfectly: `e9eb9f3c5c75710eb6cfeeb6313b5a73aaa06a28`

timhutton, 2020-01-31:
Great, thanks. That commit could usefully go in the documentation.

weigao95, 2020-02-01:
Sorry for my delayed reply as I'm struggling with the RSS deadline these days and I'm glad to see you solve the problem. I will a note to documentation.

closed 2020-02-01 by weigao95

henry123-boy, 2023-07-03:
Hi! thank you for your great work and codes. Unfortunately, I met the same problem three years latter after this issues been closed, but it seems does not work using the `e9eb9f3c5c75710eb6cfeeb6313b5a73aaa06a28` version of cilanto.