weigao95/surfelwarp #30: build failed on Ubuntu 18.04 + cuda-9 + cilantro

xiaotaw, 2020-02-18:
### OS & env
```bash
cat /etc/issue
# Ubuntu 18.04.3 LTS \n \l

gcc -v
# gcc version 6.5.0 20181026 (Ubuntu 6.5.0-2ubuntu1~18.04) 

which nvcc
# /usr/local/cuda-9.0/bin/nvcc
nvcc -V 
# nvcc: NVIDIA (R) Cuda compiler driver
# Copyright (c) 2005-2017 NVIDIA Corporation
# Built on Fri_Sep__1_21:08:03_CDT_2017
# Cuda compilation tools, release 9.0, V9.0.176

# system installed eigen 
grep -E " EIGEN_\w*_VERSION " /usr/include/eigen3/Eigen/src/Core/util/Macros.h
#define EIGEN_WORLD_VERSION 3
#define EIGEN_MAJOR_VERSION 3
#define EIGEN_MINOR_VERSION 4
```
### How to reproduce error
```
|-- data
    |-- surfelwarp
        |-- build 
        ...
    |-- Pangolin
        |-- build_static
        |-- install_static
        ...
    |-- cilantro
        |-- build_static
        |-- install_static
        ...
```
1. Building cilantro and Pangolin using system installed eigen by following https://github.com/weigao95/surfelwarp/blob/master/doc/cilantro_build.md. 

2. build
```bash
cd /data/surfelwarp

mkdir build && cd build

cmake -DCUDA_ARCH=61 \
-DVISUALIZATION_LIBRARY=cilantro \
-Dcilantro_DIR=../../cilantro/install_static/lib/cmake/cilantro \
-DPangolin_DIR=../../Pangolin/install_static/lib/cmake/Pangolin ..

make 
```
The error message is as follow: 
```bash
[  1%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o
In file included from /usr/local/cuda-9.0/include/common_functions.h:50:0,
                 from /usr/local/cuda-9.0/include/cuda_runtime.h:115,
                 from <command-line>:0:
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: error: token ""__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."" is not valid in preprocessor expressions
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: note: in definition of macro ‘__CUDACC_VER__’
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: error: token ""__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."" is not valid in preprocessor expressions
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: note: in definition of macro ‘__CUDACC_VER__’
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: error: token ""__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."" is not valid in preprocessor expressions
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: note: in definition of macro ‘__CUDACC_VER__’
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: error: token ""__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."" is not valid in preprocessor expressions
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: note: in definition of macro ‘__CUDACC_VER__’
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: error: token ""__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."" is not valid in preprocessor expressions
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^
/usr/local/cuda-9.0/include/crt/common_functions.h:64:24: note: in definition of macro ‘__CUDACC_VER__’
 #define __CUDACC_VER__ "__CUDACC_VER__ is no longer supported.  Use __CUDACC_VER_MAJOR__, __CUDACC_VER_MINOR__, and __CUDACC_VER_BUILD__ instead."
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at common_lib_generated_sanity_check.cu.o.cmake:219 (message):
  Error generating
  /data/surfelwarp/build/common/CMakeFiles/common_lib.dir//./common_lib_generated_sanity_check.cu.o

common/CMakeFiles/common_lib.dir/build.make:91: recipe for target 'common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o' failed
make[2]: *** [common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o] Error 1
CMakeFiles/Makefile2:103: recipe for target 'common/CMakeFiles/common_lib.dir/all' failed
make[1]: *** [common/CMakeFiles/common_lib.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

```

weigao95, 2020-02-19:
I guess the problem is with either Eigen or boost. Can you try a simpler program that includes both of them and make sure it works? You might need to inspect cmake cache to ensure you are testing the Eigen/boost that produces the error you mentioned.

xiaotaw, 2020-02-19:
Thank you for reply, @weigao95 . I created a simple demo, which includes eigen, cuda, and boost. The demo runs well without any errors. The code is published at  https://github.com/xiaotaw/test_for_surfelwarp/tree/master.

Any suggestions? Thank you!

weigao95, 2020-02-19:
You might need to include Eigen and boost headers in your `useCUDA.cu` file to test whether it works. The log you posted suggests the problem are with the nvcc compiler.

xiaotaw, 2020-02-20:
Thank you very much, @weigao95 !

When adding Eigen header in useCUDA.cu, the the demo failed to build. 

After a series of  test, some conclusions are:
1. system installed Eigen(3.3.4) is **NOT** compatible with CUDA(9.0.176). 
2. surfelwarp/external/eigen3.4(3.3.90) is compatible with CUDA(9.0.176)
3. By adding one line in surfelwarp/CMakeList.txt, the problem was solved:
```
@@ -78,6 +78,7 @@ set(CMAKE_DEBUG_POSTFIX d)
 # Do not use the system Eigen
 set(EIGEN_ROOT "${CMAKE_SOURCE_DIR}/external/eigen3.4")
 set(EIGEN_INCLUDE_DIR ${EIGEN_ROOT})
+include_directories(${EIGEN_INCLUDE_DIR})
 ```

closed 2020-02-20 by xiaotaw