weigao95/surfelwarp #31: error: identifier "Matrix3f" is undefined

liyiming001, 2020-03-02:
Dear author:
       I have success to cmake the file by following the second installation guide.However something error occurred.as the following:
      /surfelwarp/common/common_types.h(26): error: identifier "Matrix3f" is undefined
      /surfelwarp/common/common_types.h(26): error: expected a ";"
     ...
    and I guess that the version of eigen or boost is incompatible with project.try some version but still error.I don't know how to solve this problem.So can you give me some advice if possible.Thanks.

weigao95, 2020-03-06:
Can you pose the log from cmake?

liyiming001, 2020-03-07:
As following

liyiming001, 2020-03-07:
/home/lym/图片/1.png
/home/lym/图片/2.png
/home/lym/图片/3.png

liyiming001, 2020-03-07:
just like this

liyiming001, 2020-03-07:
just like this

liyiming001, 2020-03-07:
[image]
[image]
[image]
just like this

weigao95, 2020-03-07:
It seems the problem comes from using Eigen with CUDA. Can you try a simpler program that includes Eigen header in .cu file and make sure it works? You might need to inspect cmake cache to ensure you are testing the Eigen that produces the error you mentioned.

liyiming001, 2020-03-16:
It required Eigen3.3.9 according to your documents when install cilantro.however the latest version is 3.3.7。In the CmakeList.txt:
# Do not use the system Eigen
set(EIGEN_ROOT "${CMAKE_SOURCE_DIR}/external/eigen3.4")
set(EIGEN_INCLUDE_DIR ${EIGEN_ROOT})

I think maybe this is season why I failed. So what is the true version in your task.

closed 2020-03-22 by weigao95

Cartucho, 2020-04-21:
I had the same issue and I fixed it by uncommenting the following line: https://github.com/weigao95/surfelwarp/blob/6e547e6b33b49d903475b869a58c9bda7b9c3061/CMakeLists.txt#L70