weigao95/surfelwarp #32: Eigen3.4  (declared implicitly) cannot be referenced -- it is a deleted function

jimzou, 2020-03-03:
**My EVN:**
Visualization_library=cilantro
VS2017
CUDA10.0
Compiled Pangolin with Eigen3.4
Compiled cilantro with Eigen3.4

When I compile surfelwarp project, get error:
`1>D:/dev3rdparty/cilantro/install/lib/cmake/cilantro/../../../include\cilantro/image_point_cloud_conversions.hpp(409): **error** : function "Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::operator=(const Eigen::Block<Eigen::Matrix<float, 3, -1, 0, 3, -1>, 3, 1, true> &) [with XprType=Eigen::Matrix<float, 3, -1, 0, 3, -1>, BlockRows=3, BlockCols=1, InnerPanel=true]" (declared implicitly) cannot be referenced -- it is a deleted function
1>
1>D:/dev3rdparty/cilantro/install/lib/cmake/cilantro/../../../include\cilantro/image_point_cloud_conversions.hpp(472): **error** : function "Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::operator=(const Eigen::Block<Eigen::Matrix<float, 3, -1, 0, 3, -1>, 3, 1, true> &) [with XprType=Eigen::Matrix<float, 3, -1, 0, 3, -1>, BlockRows=3, BlockCols=1, InnerPanel=true]" (declared implicitly) cannot be referenced -- it is a deleted function
1>
1>D:/dev3rdparty/cilantro/install/lib/cmake/cilantro/../../../include\cilantro/image_point_cloud_conversions.hpp(583): **error** : function "Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::operator=(const Eigen::Block<Eigen::Matrix<float, 3, -1, 0, 3, -1>, 3, 1, true> &) [with XprType=Eigen::Matrix<float, 3, -1, 0, 3, -1>, BlockRows=3, BlockCols=1, InnerPanel=true]" (declared implicitly) cannot be referenced -- it is a deleted function
1>
1>D:/dev3rdparty/cilantro/install/lib/cmake/cilantro/../../../include\cilantro/image_point_cloud_conversions.hpp(697): **error** : function "Eigen::Block<XprType, BlockRows, BlockCols, InnerPanel>::operator=(const Eigen::Block<Eigen::Matrix<float, 3, -1, 0, 3, -1>, 3, 1, true> &) [with XprType=Eigen::Matrix<float, 3, -1, 0, 3, -1>, BlockRows=3, BlockCols=1, InnerPanel=true]" (declared implicitly) cannot be referenced -- it is a deleted function
1>
1>4 errors detected in the compilation of "C:/Users/F47/AppData/Local/Temp/tmpxft_00003c08_00000000-8_data_transfer.cpp4.ii".
1>data_transfer.cu
1>CMake Error at common_lib_generated_data_transfer.cu.obj.Release.cmake:280 (message):
1>  Error generating file
1>  D:/0.unity_prj/surfelwarp/build/common/CMakeFiles/common_lib.dir//Release/common_lib_generated_data_transfer.cu.obj`

Algomorph, 2020-03-04:
@jimzou I've commented in https://github.com/kzampog/cilantro/issues/48

closed 2020-03-06 by jimzou

mrsaleh, 2021-04-06:
Same here, Visual Studio 2019, Cuda 10.2, Eigen 3.4.