weigao95/surfelwarp #33: [Question] Does surfelwarp assume that the camera always stay at coordinate origin?

xiaotaw, 2020-04-02:
In VolumnForm dataset, the camera barely move. And I haven't found any code about coordinate transformation between camera and world.

Does surfelwarp assume that the camera always stay still, and not turn around?

weigao95, 2020-04-08:
No. There is a rigid pre-solve before the non-rigid registration.

xiaotaw, 2020-04-09:
I fetched color and depth image from a kinect dk camera (by rewriting some functions: FetchXxxxxImage). And start surfelwarp app to reconstruct a static scene(a dinner table). I held the camera, walking around the table, and the reconstruction failed.

weigao95, 2020-04-09:
The patch collider is not trained for that scene I think.

closed 2020-04-12 by weigao95

xiaotaw, 2020-05-07:
What can I do to finetune gpc model for custom scene?