weigao95/surfelwarp #35: No package 'eigen3' found

Cartucho, 2020-04-23:
When I compile the code I get the following message `No package 'eigen3' found`, which to me makes no sense since the module is in the repo under `/external/eigen3.4`

```
tribta@tribta-X550JX:~/dev/surfelwarp/build$ cmake .. -DCUDA_ARCH="70"
-- Checking for module 'eigen3'
--   No package 'eigen3' found
-- Found Boost: /usr/include (found suitable version "1.58.0", minimum required is "1.40.0") found components: system filesystem thread date_time iostreams serialization chrono atomic regex 
-- Checking for module 'libopenni'
--   No package 'libopenni' found
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing: OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** io features related to openni2 will be disabled
-- Could NOT find ensenso (missing: ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing: _DSSDK_LIBRARIES) 
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

** WARNING ** io features related to libusb-1.0 will be disabled
-- Checking for module 'flann'
--   No package 'flann' found
-- Checking for module 'flann'
--   No package 'flann' found
-- Checking for module 'libopenni'
--   No package 'libopenni' found
-- Checking for module 'libopenni2'
--   No package 'libopenni2' found
-- Could NOT find OpenNI2 (missing: OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
** WARNING ** visualization features related to openni2 will be disabled
-- Could NOT find ensenso (missing: ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** visualization features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing: DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** visualization features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing: _DSSDK_LIBRARIES) 
** WARNING ** visualization features related to dssdk will be disabled
-- Could NOT find RSSDK (missing: _RSSDK_LIBRARIES) 
** WARNING ** visualization features related to rssdk will be disabled
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- looking for PCL_KDTREE
-- looking for PCL_SEARCH
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_2D
-- looking for PCL_GEOMETRY
-- looking for PCL_FEATURES
-- looking for PCL_ML
-- looking for PCL_SEGMENTATION
-- looking for PCL_VISUALIZATION
-- looking for PCL_SURFACE
-- looking for PCL_REGISTRATION
-- looking for PCL_KEYPOINTS
-- looking for PCL_TRACKING
-- looking for PCL_RECOGNITION
-- looking for PCL_STEREO
-- looking for PCL_OUTOFCORE
-- looking for PCL_PEOPLE
-- Found Boost: /usr/include (found version "1.58.0") found components: system filesystem 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tribta/dev/surfelwarp/build
```

weigao95, 2020-04-23:
It seems the build file generation succeeds. What happened if you just type `make`.

Cartucho, 2020-04-24:
If I type `make` I get the following error:
I checked and it seems to be the same error as (#31). Do you think the problem has to do with the CUDA 9.0 version? Which version did you use, 9.2?

```
tribta@tribta-X550JX:~/dev/surfelwarp/build$ make
[  1%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o
/home/tribta/dev/surfelwarp/common/common_types.h(25): error: identifier "Matrix3f" is undefined

/home/tribta/dev/surfelwarp/common/common_types.h(25): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(26): error: identifier "Vector3f" is undefined

/home/tribta/dev/surfelwarp/common/common_types.h(26): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(27): error: identifier "Matrix4f" is undefined

/home/tribta/dev/surfelwarp/common/common_types.h(27): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(28): error: identifier "Vector4f" is undefined

/home/tribta/dev/surfelwarp/common/common_types.h(28): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(29): error: identifier "Matrix6f" is undefined

/home/tribta/dev/surfelwarp/common/common_types.h(29): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(30): error: identifier "Vector6f" is undefined

/home/tribta/dev/surfelwarp/common/common_types.h(30): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(31): error: identifier "MatrixXf" is undefined

/home/tribta/dev/surfelwarp/common/common_types.h(31): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(32): error: identifier "VectorXf" is undefined

/home/tribta/dev/surfelwarp/common/common_types.h(32): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(33): error: identifier "Isometry3f" is undefined

/home/tribta/dev/surfelwarp/common/common_types.h(33): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(38): error: expected a declaration

/home/tribta/dev/surfelwarp/common/common_types.h(38): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(41): error: expected a declaration

/home/tribta/dev/surfelwarp/common/common_types.h(41): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(46): error: expected a declaration

/home/tribta/dev/surfelwarp/common/common_types.h(46): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(49): error: expected a declaration

/home/tribta/dev/surfelwarp/common/common_types.h(49): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(52): error: expected a declaration

/home/tribta/dev/surfelwarp/common/common_types.h(52): error: expected a ";"

/home/tribta/dev/surfelwarp/common/common_types.h(55): error: expected a declaration

/home/tribta/dev/surfelwarp/common/common_types.h(55): error: expected a ";"

/home/tribta/dev/surfelwarp/common/logging.h(23): error: identifier "nullptr" is undefined

/home/tribta/dev/surfelwarp/common/logging.h(26): error: identifier "nullptr" is undefined

[... 156 lines cut ...]

/home/tribta/dev/surfelwarp/common/sanity_check.cu(76): error: explicit type is missing ("int" assumed)

/home/tribta/dev/surfelwarp/common/sanity_check.cu(83): error: explicit type is missing ("int" assumed)

/home/tribta/dev/surfelwarp/common/sanity_check.cu(89): error: identifier "DeviceArray" is undefined

/home/tribta/dev/surfelwarp/common/sanity_check.cu(89): error: type name is not allowed

/home/tribta/dev/surfelwarp/common/sanity_check.cu(89): error: identifier "key_in" is undefined

Error limit reached.
100 errors detected in the compilation of "/tmp/tmpxft_00001ac9_00000000-6_sanity_check.cpp1.ii".
Compilation terminated.
CMake Error at common_lib_generated_sanity_check.cu.o.cmake:266 (message):
  Error generating file
  /home/tribta/dev/surfelwarp/build/common/CMakeFiles/common_lib.dir//./common_lib_generated_sanity_check.cu.o

common/CMakeFiles/common_lib.dir/build.make:3404: recipe for target 'common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o' failed
make[2]: *** [common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o] Error 1
CMakeFiles/Makefile2:103: recipe for target 'common/CMakeFiles/common_lib.dir/all' failed
make[1]: *** [common/CMakeFiles/common_lib.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
```

weigao95, 2020-04-27:
Yeah, that seems to be the probem with Eigen and CUDA. Can you try to debug it with a simpler program that includes Eigen header in .cu file and make sure it works? You might need to inspect cmake cache to ensure you are testing the Eigen that produces the error you mentioned.

closed 2020-05-02 by weigao95