weigao95/surfelwarp #37: support online rendering

xiaotaw, 2020-04-26:
Hi, I noticed surfelwarp support offline rendering. It's very cool to display rendered images (e.g. live_abedo.png) on screen, instead of saving into local files. 

Simply comment out `glfwWindowHint(GLFW_VISIBLE, GL_FALSE);`, it just displays a black window. HAHA

Thank you!

weigao95, 2020-04-27:
You need to delete the `glBindFramebuffer` in the drawing function. Here is a tutorial: https://learnopengl.com/Advanced-OpenGL/Framebuffers

xiaotaw, 2020-04-29:
Done, Thx!

closed 2020-04-29 by xiaotaw

daxiong-123, 2020-05-06:
Hello, could you tell me which `glBindFramebuffer` should I delete? @xiaotaw

xiaotaw, 2020-05-07:
> Hello, could you tell me which `glBindFramebuffer` should I delete? @xiaotaw

Always set the second parameter to be 0, and the pics/images will be updated in windows.
glBindFramebuffer(GL_FRAMEBUFFER, 0);  

Note: Based on the current code, you also have to set 'offline rendering' with True.

daxiong-123, 2020-05-09:
> > Hello, could you tell me which `glBindFramebuffer` should I delete? @xiaotaw
> 
> Always set the second parameter to be 0, and the pics/images will be updated in windows.
> glBindFramebuffer(GL_FRAMEBUFFER, 0);
> 
> Note: Based on the current code, you also have to set 'offline rendering' with True.

Thanks, i tried but still failed. Maybe I should look through the whole project.

xiaotaw, 2020-05-09:
@daxiong-123 try one more time? it should work:
1. comment out glfwWindowHint(GLFW_VISIBLE, GL_FALSE) to make sure the blank window appears
2. find function `drawVisualizationMap` in file RendererDraw.cpp, and 
modify glBindFramebuffer ( line 191)
3. in the same function `drawVisualizationMap`, insert `glfwSwapBuffers(mGLFWwindow);` before cleanup code (line 218).

411066282, 2021-09-02:
@xiaotaw dear,   could you share your whole modified code project to me? I just failed this, thank you very much!                      yaoyao-501@163.com

xiaotaw, 2021-09-14:
> @xiaotaw dear, could you share your whole modified code project to me? I just failed this, thank you very much! [yaoyao-501@163.com](mailto:yaoyao-501@163.com)

Here is my modified code. Hope its commit log would help.  https://github.com/xiaotaw/surfelwarp

411066282, 2021-10-15:
@xiaotaw thank you very much! you are one  good and kind guy!!!