weigao95/surfelwarp #38: fatal error: pcl/visualization/pcl_visualizer.h: No such file or directory

Cartucho, 2020-04-29:
I decided to try the compilation in another workstation with Ubuntu 16.04 and CUDA 10.2. However, I get the following error `fatal error: pcl/visualization/pcl_visualizer.h: No such file or directory`.

I added to the CMakeLists.txt the following lines:
```
set(CUDA_ARCH "75")
set(VISUALIZATION_LIBRARY=cilantro)
set(cilantro_DIR "/home/tribta/dev/cilantro")
```
and I uncommented line 54 
`    set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")`

Then, I tried to compile:

```
tribta@samt-HP-Z8-G4-Workstation:~/dev/surfelwarp/build$ cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-10.2 (found suitable version "10.2", minimum required is "9") 
-- Checking for module 'eigen3'
--   Found eigen3, version 3.2.92
CMake Warning (dev) at /opt/cmake-3.17.1-Linux-x86_64/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (eigen) does
  not match the name of the calling package (PCL).  This can lead to problems
  in calling code that expects `find_package` result variables (e.g.,
  `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/share/pcl-1.8/PCLConfig.cmake:146 (find_package_handle_standard_args)
  /usr/local/share/pcl-1.8/PCLConfig.cmake:631 (find_eigen)
  /usr/local/share/pcl-1.8/PCLConfig.cmake:836 (find_external_library)
  CMakeLists.txt:96 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found eigen: /usr/include/eigen3  
-- Found Boost: /home/tribta/dev/libboost/installation/boost_1_72_0 (found suitable version "1.72.0", minimum required is "1.40.0") found components: system filesystem thread date_time iostreams chrono atomic regex 
-- Checking for module 'libopenni'
--   Found libopenni, version 1.5.4.0
CMake Warning (dev) at /opt/cmake-3.17.1-Linux-x86_64/share/cmake-3.17/Modules/FindPackageHandleStandardArgs.cmake:272 (message):
  The package name passed to `find_package_handle_standard_args` (openni)
  does not match the name of the calling package (PCL).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/share/pcl-1.8/PCLConfig.cmake:228 (find_package_handle_standard_args)
  /usr/local/share/pcl-1.8/PCLConfig.cmake:637 (find_openni)
  /usr/local/share/pcl-1.8/PCLConfig.cmake:833 (find_external_library)
  CMakeLists.txt:96 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found openni: /usr/lib/libOpenNI.so  
[... 600 lines cut ...]

/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/stitching/detail/warpers.hpp(213): warning: overloaded virtual function "cv::detail::PlaneWarper::buildMaps" is only partially overridden in class "cv::detail::AffineWarper"

/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/stitching/detail/warpers.hpp(213): warning: overloaded virtual function "cv::detail::PlaneWarper::warp" is only partially overridden in class "cv::detail::AffineWarper"

/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/videostab/motion_stabilizing.hpp(106): warning: function "cv::videostab::IMotionStabilizer::stabilize(int, const std::vector<cv::Mat, std::allocator<cv::Mat>> &, std::pair<int, int>, cv::Mat *)" is hidden by "cv::videostab::GaussianMotionFilter::stabilize" -- virtual function override intended?

/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/stitching/detail/warpers.hpp(213): warning: overloaded virtual function "cv::detail::PlaneWarper::buildMaps" is only partially overridden in class "cv::detail::AffineWarper"

/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/stitching/detail/warpers.hpp(213): warning: overloaded virtual function "cv::detail::PlaneWarper::warp" is only partially overridden in class "cv::detail::AffineWarper"

/opt/ros/kinetic/include/opencv-3.3.1-dev/opencv2/videostab/motion_stabilizing.hpp(106): warning: function "cv::videostab::IMotionStabilizer::stabilize(int, const std::vector<cv::Mat, std::allocator<cv::Mat>> &, std::pair<int, int>, cv::Mat *)" is hidden by "cv::videostab::GaussianMotionFilter::stabilize" -- virtual function override intended?

Scanning dependencies of target visualization_lib
[ 21%] Building CXX object visualization/CMakeFiles/visualization_lib.dir/ImageVisualizer.cpp.o
[ 22%] Building CXX object visualization/CMakeFiles/visualization_lib.dir/PointCloudVisualizer.cpp.o
/home/tribta/dev/surfelwarp/visualization/PointCloudVisualizer.cpp:14:46: fatal error: pcl/visualization/pcl_visualizer.h: No such file or directory
compilation terminated.
visualization/CMakeFiles/visualization_lib.dir/build.make:102: recipe for target 'visualization/CMakeFiles/visualization_lib.dir/PointCloudVisualizer.cpp.o' failed
make[2]: *** [visualization/CMakeFiles/visualization_lib.dir/PointCloudVisualizer.cpp.o] Error 1
CMakeFiles/Makefile2:512: recipe for target 'visualization/CMakeFiles/visualization_lib.dir/all' failed
make[1]: *** [visualization/CMakeFiles/visualization_lib.dir/all] Error 2
Makefile:103: recipe for target 'all' failed
make: *** [all] Error 2
```

weigao95, 2020-04-29:
Check your cmake cache to see if `WITH_PCL` is ON.

It seems you are compiling with `cilantro`. Then a pretty hacky solution would be just comment the pcl headers.

Cartucho, 2020-04-29:
Yes, I am using cilantro as instructed in the README since I am using `CUDA 10`.

I checked and there was no `WITH_PCL` parameter in the "CMakeCache.txt" file.

In which files should I comment out the PCL headers?

weigao95, 2020-05-02:
The pcl header in `visualization/PointCloudVisualizer.cpp`

Cartucho, 2020-05-04:
I believe the error was being caused since I set wrongly the libraries in the CMake. I am not 100% sure but I think I should have done:

```
set(CUDA_ARCH "75")
# For CUDA 10
set(VISUALIZATION_LIBRARY "cilantro")
set(cilantro_DIR "/home/tribta/dev/cilantro")
set(Pangolin_DIR "/home/tribta/dev/Pangolin")
```

closed 2020-05-04 by Cartucho