weigao95/surfelwarp #41: make

shugao1, 2020-05-03:
gyc@3Dre:~/桌面/3D/surfelwarp/build$ cmake ..
-- Found OpenMP_C: -fopenmp  
-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE   
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gyc/桌面/3D/surfelwarp/build
gyc@3Dre:~/桌面/3D/surfelwarp/build$ make
[  1%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o
cc: error trying to exec 'cc1plus': execvp: 没有那个文件或目录
CMake Error at common_lib_generated_sanity_check.cu.o.cmake:220 (message):
  Error generating
  /home/gyc/桌面/3D/surfelwarp/build/common/CMakeFiles/common_lib.dir//./common_lib_generated_sanity_check.cu.o

common/CMakeFiles/common_lib.dir/build.make:108: recipe for target 'common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o' failed
make[2]: *** [common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o] Error 1
CMakeFiles/Makefile2:337: recipe for target 'common/CMakeFiles/common_lib.dir/all' failed
make[1]: *** [common/CMakeFiles/common_lib.dir/all] Error 2
Makefile:100: recipe for target 'all' failed
make: *** [all] Error 2

shugao1, 2020-05-03:
> gyc@3Dre:~/桌面/3D/surfelwarp/build$ cmake .. -- Found OpenMP_C: -fopenmp -- Found OpenMP_CXX: -fopenmp -- Found OpenMP: TRUE -- Configuring done -- Generating done -- Build files have been written [...]

closed 2020-05-03 by shugao1

wangmiaowei, 2020-12-01:
Have you solved this error? I am also meeting it and really need help.