weigao95/surfelwarp #42: When I tried 'make' in surfelwarp

shugao1, 2020-05-03:
I have installed eigen3.3.9 pangolin cilantro in /usr/local/include, and the configuration of cmake surfelwarp has been successfully configured, but make error:

gyc@3Dre:~/Desktop/3D/surfelwarp/build$ make
[  1%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_sanity_check.cu.o
/home/gyc/Desktop/3D/surfelwarp/common/logging.h: In destructor ‘surfelwarp::LogMessageFatal::~LogMessageFatal()’:
/home/gyc/Desktop/3D/surfelwarp/common/logging.h:114:49: warning: throw will always call terminate() [-Wterminate]
    throw new std::runtime_error(log_stream_.str());
                                                 ^
/home/gyc/Desktop/3D/surfelwarp/common/logging.h:114:49: note: in C++11 destructors default to noexcept
[  1%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_algorithm_types.cu.o
[  2%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_common_texture_utils.cu.o
/home/gyc/Desktop/3D/surfelwarp/common/logging.h: In destructor ‘surfelwarp::LogMessageFatal::~LogMessageFatal()’:
/home/gyc/Desktop/3D/surfelwarp/common/logging.h:114:49: warning: throw will always call terminate() [-Wterminate]
    throw new std::runtime_error(log_stream_.str());
                                                 ^
/home/gyc/Desktop/3D/surfelwarp/common/logging.h:114:49: note: in C++11 destructors default to noexcept
[  2%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_common_utils.cu.o
[  3%] Building NVCC (Device) object common/CMakeFiles/common_lib.dir/common_lib_generated_data_transfer.cu.o
In file included from /home/gyc/Desktop/3D/surfelwarp/common/data_transfer.h:8:0,
                 from /home/gyc/Desktop/3D/surfelwarp/common/data_transfer.cu:1:
/home/gyc/Desktop/3D/surfelwarp/common/point_cloud_typedefs.h:13:36: fatal error: cilantro/point_cloud.hpp: 没有那个文件或目录
 #include <cilantro/point_cloud.hpp>
                                    ^
compilation terminated.
CMake Error at common_lib_generated_data_transfer.cu.o.cmake:220 (message):
  Error generating
  /home/gyc/Desktop/3D/surfelwarp/build/common/CMakeFiles/common_lib.dir//./common_lib_generated_data_transfer.cu.o

common/CMakeFiles/common_lib.dir/build.make:101: recipe for target 'common/CMakeFiles/common_lib.dir/common_lib_generated_data_transfer.cu.o' failed
make[2]: *** [common/CMakeFiles/common_lib.dir/common_lib_generated_data_transfer.cu.o] Error 1
CMakeFiles/Makefile2:337: recipe for target 'common/CMakeFiles/common_lib.dir/all' failed
make[1]: *** [common/CMakeFiles/common_lib.dir/all] Error 2
Makefile:100: recipe for target 'all' failed
make: *** [all] Error 2

[image]

closed 2020-05-04 by shugao1

wangmiaowei, 2020-12-01:
Have you solved this problem? I also meet it.

shugao1, 2020-12-01:
The file reference path in the code file is incorrect. Use the &quot;locate file name&quot; to find the file path, such as 'locate point_cloud.hpp ', and finally change the code.------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;&quot;wangmiaowei&quot;<notifications@github.com&gt;
发送时间:&nbsp;2020年12月1日(星期二) 上午10:40
收件人:&nbsp;&quot;weigao95/surfelwarp&quot;<surfelwarp@noreply.github.com&gt;;
抄送:&nbsp;&quot;shugao1&quot;<583931277@qq.com&gt;;&quot;State change&quot;<state_change@noreply.github.com&gt;;
主题:&nbsp;Re: [weigao95/surfelwarp] When I tried &#39;make&#39; in surfelwarp (#42)