weigao95/surfelwarp #43: Failed to Solve on Boxing-like dataset

xiaotaw, 2020-05-12:
Hello, I was trying to repeat `Boxing` using a live camera. It failed to seperate fist from face.
cliped_filter_depth:
[image]
reference_phong:
[image]
live_phone:
[image]

weigao95, 2020-05-22:
You might try to modify the re-init parameter.

closed 2020-05-22 by weigao95

411066282, 2021-09-04:
@xiaotaw how did you do this, with live camera? just modify the fetch image interface?