weigao95/surfelwarp #47: Nan in node candidate error

bsespede, 2020-07-21:
Hi again @weigao95 

I finally had some time to look into the error mentioned in #34 :
... /home/bsespede/Repos/surfelwarp/apps/surfelwarp_app/main.cpp:40: The 57th Frame 
/home/bsespede/Repos/surfelwarp/core/geometry/WarpFieldUpdater.cpp:21: Nan in node candidate
terminate called after throwing an instance of 'std::runtime_error*'

As you mentioned I tried visualizing the correspondences of the crashing frame and it looks something like this:
[Image of correspondences](https://drive.google.com/file/d/1JtzOOEagHJM_mnvew03aP2gsE_nxppGJ/view?usp=sharing)
[Scene](https://drive.google.com/file/d/1EAZllbnXLPAPKzzouzN73BJNti2hiem9/view?usp=sharing)

I suspect some of the correspondences are incorrect, could you provide me some information on how to retrain the global patch collider?

Thanks again!

bsespede, 2020-07-22:
I've just tested the same scene on gpc_test and I've got a similar result:
[Correspondence son gpc_test](https://drive.google.com/file/d/1UJZ6ga8KgPBvcdkEA98RXzFFvYKeMxWu/view?usp=sharing)

closed 2020-07-22 by bsespede

henry123-boy, 2023-08-05:
Hi, I met the same problem when running the code in the example data, how did u solve this problem?