weigao95/surfelwarp #50: how to visualize the program output while running?

ByLee1997, 2020-10-11:
I've run the surfelwarp_app and it execute normally and produce png output in the folder, but I can't see the output when running.

I tried to add _**Visualizer::DrawPointCloud(observation.vertex_config_map**)_ in _**SurfelWarpSerial::ProcessNextFrameWithReinit(bool offline_save)**_, but it only show a white dot, and it stock at there, close the window cause segmentation fault. 

When I change the **_glfwWindowHint(GLFW_VISIBLE, GL_TRUE)_** in **_surfelwarp::Renderer::initGLFW()_**, it also create a window and has an error of no response.

My environment is ubuntu16.04 RTX2070 cuda10.0, using cilantro

weigao95, 2020-11-08:
See: https://github.com/weigao95/surfelwarp/issues/37#issue-606914871

closed 2020-11-13 by weigao95

411066282, 2021-09-02:
@ByLee1997 Have you solved this? After seeing "#37 (comment)"