weigao95/surfelwarp #52: 'nlohmann::detail::parse_error'   what():  [json.exception.parse_error.101] parse error at 1: syntax error  - unexpected end of input; expected '[', '{', or a literal Aborted

Wangyouai, 2020-11-03:
Ubuntu18.04、cuda10.0、gcc7.5
After my surfelwarp project make is successful, I configured the gpc and boxing paths, and ran **_./surfelwarp_app /path/to/config._** An error of'nlohmann::detail::parse_error appeared, have you encountered this situation?

weigao95, 2020-11-08:
I haven't seen it. That's sound like a parsing error of json file. Have you edited the json config?

Wangyouai, 2020-11-09:
Yes, I only modified the path of the gpc and boxing database of thesurfelwarp/test_data/boxing_config.json file through the text editor that comes with ubuntu. The modified code is as follows, and I will send you the source file as an attachment.
{

    "clip_far":1300,

    "clip_near":10,

    "data_prefix":"/home/wya/project/SW_cliantro/VolumeDeformData/boxing/data",

    "depth_focal_x":570.0,"depth_focal_y":570.0,"depth_principal_x":320.0,"depth_principal_y":240.0,

    "gpc_model_path":"/home/wya/project/SW_cliantro/gpc/sintel_small_speed",

    "image_cols":640,"image_rows":480,

    "start_frame":0,

    "num_frame":310,

    "rgb_focal_x":570.0,"rgb_focal_y":570.0,"rgb_principal_x":320.0,"rgb_principal_y":240.0,

    "use_periodic_reinit":false,

    "reinit_period":30,

    "use_density":true,

    "use_foreground":false,

    "use_offline_foreground":false

}

weigao95, 2020-11-10:
I don't see anything wrong with that, but I don't have a windows platform now for testing. Maybe you can ask the author of json.h for help as he should be the expert.

closed 2020-11-17 by weigao95

Wangyouai, 2020-11-18:
Sorry, I have been away for a while. The reason for the previous error should be that the path to open the boxing_config.json file is wrong, and an absolute path is required. Refer to this link：json.exception.parse_error
So I want to ask how to check the code that reads the json file in your project, and then see if this problem can be solved.Because when I used clion to view your project, I didn't find the code involved in reading the json file.

At the same time, according to this description IDE error. I switched to ubuntu16.04 and followed the same steps to build the project without installing other unrelated software packages, and encountered the same error when I reached the last one.

Closed #52.

Wangyouai, 2020-11-23:
My solution：modify the **_surfelwarp/apps/surfelwarp_app/main.cpp_**
```
#include "common/common_utils.h"
#include "common/ConfigParser.h"
#include "core/SurfelWarpSerial.h"
#include <boost/filesystem.hpp>

int main(int argc, char** argv) {
	using namespace surfelwarp;
	//Get the config path
	std::string config_path;
       config_path = "/home/wya/project/SW/surfelwarp/test_data/boxing_config.json";

//Parse it
auto& config = ConfigParser::Instance();
config.ParseConfig(config_path);
```