weigao95/surfelwarp #54: How to acquire deformation vector field of 2 point clouds?

qaler, 2020-11-18:
Hello wegao95! Is there any way to apply deformation registration to 2 point clouds and acquire the deformation vector field by using your library? I guess the registration result of 2 depth frames is stored inside [SurfelWarpSerial.cpp](https://github.com/weigao95/surfelwarp/blob/master/core/SurfelWarpSerial.cpp) code, but I don't know how to get the vector field from it.

Wangyouai, 2020-11-19:
Hello, are you using ubuntu and cuda10.0?

weigao95, 2020-11-29:
The registration code is tightly integrated with other parts, like image and geometry processing. I think it is not easy to separate that out. But you might convert your two point-clouds into two depth images and try to run the algorithm on it.

closed 2020-12-06 by weigao95