weigao95/surfelwarp #55: Run error:surfelwarp/common/algorithm_types.h:85

Wangyouai, 2020-11-23:
hello，I encountered a Error after running `/surfelwarp_app /path/to/config`
`wya@wya:~/project/SW/surfelwarp/build/apps/surfelwarp_app$ ./surfelwarp_app /path/to/config
Error: 	/home/wya/project/SW/surfelwarp/common/algorithm_types.h:85
`
My computer graphics card is GTX1070 , I used the cuda10.0 and cilantro, so I set the **cuda_arch=61** as follows:
I have modified the CMakeLists:
`# Cuda
find_package(CUDA 10 REQUIRED)
include_directories(${CUDA_INCLUDE_DIRS})
set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -arch=sm_61")
set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -gencode=arch=compute_61,code=sm_61")
set(CUDA_NVCC_FLAGS "${CUDA_NVCC_FLAGS} -gencode=arch=compute_61,code=compute_61")

# Point cloud library and a series / cilantro

set(cilantro_DIR "../../cilantro/install_static")
find_package(cilantro REQUIRED)
include_directories(${cilantro_INCLUDE_DIRS})
add_definitions(-DWITH_CILANTRO)
`
This is my cmake information:
`-- The C compiler identification is GNU 6.5.0
-- The CXX compiler identification is GNU 6.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc - works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ - works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda (found suitable version "10.0", minimum required is "10") 
-- Found OpenMP_C: -fopenmp (found version "4.5") 
-- Found OpenMP_CXX: -fopenmp (found version "4.5") 
-- Found OpenMP: TRUE (found version "4.5")  
-- Found Boost: /usr/include (found version "1.58.0") found components: system filesystem 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wya/project/SW/surfelwarp/cmake-build-debug
`
Have you encountered this problem?

weigao95, 2020-11-29:
I would recommend you try a simple program with `cudaMalloc` and `cudaSafeCall`, using exactly the cuda setup you mentioned to see if the environment is setup correctly.

Wangyouai, 2020-12-01:
My solution:I used ubuntu16.04 before, and its adapted GTX1070 GPU driver is 390, but cuda10.0 needs 410 or more, So I switched to ubuntu18.04, and it can run with its adapted GPU driver 450.

closed 2020-12-03 by Wangyouai