weigao95/surfelwarp #56: incompatible for higher-level cuda & gpu

Cryst4L9527, 2020-12-29:
when I try to run this code on nividia RTX3090 with cuda11.1,there is a new problem as below: 
'shfl' without '.sync' is not supported on .target sm_70 and higher from PTX ISA version 6.4 
It seems that the cuda code is not suitable for the novel gpu for some reason.Would you update your code to fit the new feature?I'll appreciate it very much!

BaldrLector, 2021-01-20:
Hi, you need to change the CMakeLitsts.txt, refer to #23

Cryst4L9527, 2021-01-31:
I've changed the CMakeLists.txt，it seems that the problem is the cuda10+ doesn't support the shfl without sync anymore.

weigao95, 2021-02-04:
I remember that shfl will be deprecated. You need to change the device instrinsics. In the `common/device_instrinsics.h` file, you need to change `shfl.up.b32` into `shfl.sync.up.b32 ` according to this doc: https://docs.nvidia.com/cuda/parallel-thread-execution/index.html. You need to add `sync` for all data types.

Cryst4L9527, 2021-02-05:
Thank you very much！In fact I change the arch to sm_61 when compile and it worked too，but maybe this operation will affect the speed.I'll try to change them.

closed 2021-02-06 by weigao95