weigao95/surfelwarp #62: The pre-allocated buffer is not enough

yhlong666, 2021-02-21:
Hi,

When I run the program the error happened:

> DeviceBufferArray.h:90: The pre-allocated buffer is not enough 
> terminate called after throwing an instance of 'std::runtime_error*'
> Aborted (core dumped)

I try to incresea the max node number defined in the global_configs.h,
But another error happened when I tried to compile the program:

How to solve this?

ptxas error   : File uses too much global constant data (0x10010 bytes, 0x10000 max)
CMake Error at core_lib_generated_KNNBruteForceLiveNodes.cu.o.cmake:280 (message):
  Error generating file
/surfelwarp/build/core/CMakeFiles/core_lib.dir/geometry/./core_lib_generated_KNNBruteForceLiveNodes.cu.o

Thanks a lot.

weigao95, 2021-02-27:
Can you clarify which buffer is not large enough? You might need to set the `CUDA_DEBUG_SYNC_CHECK ` flag in `common/global_configs.h` to see where the exception is thrown.

closed 2021-03-03 by weigao95

vhvkhoa, 2021-04-30:
> Can you clarify which buffer is not large enough? You might need to set the `CUDA_DEBUG_SYNC_CHECK ` flag in `common/global_configs.h` to see where the exception is thrown.

Hello Wei Gao, I followed your instruction to turn on the `CUDA_DEBUG_SYNC_CHECK` flag but I got this error while re-compiling:
```
make[2]: *** [common/CMakeFiles/common_lib.dir/build.make:2996: common/CMakeFiles/common_lib.dir/common_lib_generated_data_transfer.cu.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:339: common/CMakeFiles/common_lib.dir/all] Error 2
make: *** [Makefile:103: all] Error 2
```
Can I get some suggestions from you to work this out ?
Thank you a lot !