weigao95/surfelwarp #64: /surfelwarp/core/render/render.cpp:67: The GLFW window is not correctly created.

ZYJxxx, 2021-03-25:
when i run "./surfelwarp_app /path/to/config", there is the error. I can normal run [https://www.glfw.org/docs/latest/quick.html](url),  my computer has 2 TiTan V GPU's , and -DCUDA_ARCH=70 was set when i cmake this project. 
I have no idea about this now, looking forward for your reply.

weigao95, 2021-03-29:
Please try again with the same setup in https://github.com/weigao95/surfelwarp/blob/master/core/render/Renderer.cpp

There might be some flags not supported.

ZYJxxx, 2021-03-30:
> There might be some flags not supported.

Thank you for your reply, i found i didn't use the external video card's opengl...

closed 2021-03-30 by ZYJxxx

LiuXinqi12, 2021-10-02:
@ZYJxxx  Hi! Have you solved this problme?