weigao95/surfelwarp #65: Headless mode

yuehaowang, 2021-03-25:
Dear authors,

Thanks for providing such an awesome project. I would like to run surfelwarp app on my server without GUI and display (just want to perform reconstruction and output results to files). However, the code will always run glfw and OpenGL, which leads to runtime errors on my server. I am wondering if there is any workaround to disable glfw and OpenGL, i.e., run the reconstruction program in a headless mode?

Thank you very much in advance for any help.

weigao95, 2021-03-29:
OpenGL plays a very important role in reconstruction and it is almost impossible to remove it. See https://github.com/weigao95/surfelwarp/issues/16#issuecomment-554322225.

closed 2021-04-01 by weigao95

shaisung, 2022-06-14:
There's a GLFW branch with headless EGL support:

https://github.com/glfw/glfw/pull/1608

A bit hacky but it did the trick for me.

There's also various methods of setting up a docker with VirtualGL, I haven't been able to set that up yet but it might work as well.
Examples:
https://github.com/ffeldhaus/docker-xpra-html5-gpu-minimal
https://github.com/dcommander/virtualgl_docker_examples