weigao95/surfelwarp #66: Windows Prebuilt binary doesn't seem right

mrsaleh, 2021-04-03:
Hello
The link to windows binary doesn't point to a valid executable binary for the Windows platform.
The file extension is not .exe and when I rename it to .exe, I can't run it.

weigao95, 2021-04-28:
You need to run it in the command line (or maybe windows power shell?). Navigate to the path that contains the app and run './surfelwarp_app'.

mrsaleh, 2021-05-03:
The issue is that this is not a windows executable file.

hwy1992129, 2021-07-08:
> You need to run it in the command line (or maybe windows power shell?). Navigate to the path that contains the app and run './surfelwarp_app'.

I can't run it in windows, either.
Add ".exe" and run it in PowerShell, it said, "The specified executable is not a valid application for this OS platform"... Without ".exe", it will pop up a window, "How do you want to open the file", which is looking for a default app to open it.
In windows system, open a linux shell to run it. It says, "./surfelwarp_app: error while loading shared libraries: libvtkRenderingLOD-6.2.so.6.2: cannot open shared object file: No such file or directory", which is the same as I run in Ubuntu.....

closed 2021-09-01 by weigao95