weigao95/surfelwarp #67: Confused about the flip_color parameter.

yuehaowang, 2021-04-22:
Hi, I am recently studying the code of surfelwarp. I try to save the colored point clouds but it seems the RGBs are saved in a reversed order, i.e., BGR.  I also noticed the `downloadColoredPointCloud` function is defined as

```cpp
PointCloud3fRGB_Pointer downloadColoredPointCloud(
	cudaTextureObject_t vertex_map,
	cudaTextureObject_t color_time_map,
	bool flip_color = false
);
```

where the `flip_color` parameter will determine whether the color of point clouds will be flipped.

I am confused about why this parameter is needed and why the color of point clouds is reversed. Is it caused by the endianness of the underlying architecture?

Thank you very much for your help in advance.

weigao95, 2021-04-28:
If I remember correctly (which maybe not), this is because the color encoding in OpenCV is 'BGR' instead of 'RGB'.

yuehaowang, 2021-05-03:
Thanks! Got it.

closed 2021-05-03 by yuehaowang