weigao95/surfelwarp #70: What is the purpose of using GPC in Surfelwarp ?

vhvkhoa, 2021-05-11:
Hello,

I am kind of new in 3D reconstruction and your work is very impressive to me, the source code is very well organized.
I have successfully run the code on my own recorded videos and obtained very promising results.

Currently, I am trying to understand the correspondence pairs detector in Surfelwarp, how it (the set $P$ in your paper) is produced to find a way of improving your work on my dataset. However, I found there was no reference of GPC in the paper.
I am not sure what is its true function in the whole pipeline of Surfelwarp, I've also read through a related work of K. Guo et al. (which is referred to in the paper for producing $P$) but they seemed to use the ICP method to find $P$ ?

I have used your apps/gpc_test/main.cpp to visualize the correspondence pairs with some modification to show all the pairs. And it does not make much sense to me because there is no correspondence pair in some parts but Surfelwarp is still able to produce points at those parts.
For example, the image below shows no pairs at the legs but reconstructed object still well captures the legs.
[image]

vhvkhoa, 2021-05-22:
Hello,
After reading through the source code, my understanding is that you are using correspondence pairs to enforce the deformation graph, which is not in the original paper of SurfelWarp. However, I am not sure on the final energy function after updating this new term. Can you show me the new energy function ?

weigao95, 2021-09-01:
Sorry that I am working on my thesis these days. The GPC is not described in the original paper, and our original implementation doesn't include this term. For the energy function, you may read the fusion4d paper.

closed 2021-09-01 by weigao95

vhvkhoa, 2021-09-01:
Thank you for your answer. Could you please give me a hint on where you define the energy function to be optimized in your codebase ?

weigao95, 2021-09-02:
`DenseDeepHandler.h`, `DensityForegroundHandler.h`, `NodeGraphSmoothHandler.h`, `SparseCorrespondenceHandler.h` in `core/warp_solver`